CodethinkLabs/mt3620-m4-samples #1: The actual output does not match exactly with the Readme.md's output.

Shieroo, 2020-02-13:
**Issue description:**
The output of 'CodeLinkLabs_UART_RTApp_MT3620_BareMetal''s Readme.md does not match exactly with the actual result. 

**Actual Results:** 
The data is receiving separated 3 lines as 12 bytes and 10 bytes with line breaks.

For example,
UART received 12 bytes: 'RTCore: Hell'.
UART received 10 bytes: 'o world!
'.

**Expected result** per the Readme.md.
The output  displays 22 bytes with a string line without break from the Readme.md of the sample.

aiden-ct, 2020-02-14:
Resolved in ea1440cc514845ff0b81972b9dd93a20dc8c22ee.

closed 2020-02-14 by aiden-ct

aiden-ct, 2020-02-14:
Note that there is no guarantee on the output splitting as it does. The non-debug UARTs use DMA, which use a variable size buffer. This is why the word "similar" was used in the README.