CodethinkLabs/mt3620-m4-samples #5: SPI sample shows 5v connection to ST_MK1160V1 should be 3.3v

n7jti, 2020-05-01:
In the Connection Diagram for SPI sample shows both VDD and VDDIO of the ST_MK1160V1 connected to H3 Pin 1 of the Seeed Board. H3 Pin 3 is documented as being 5v. See: https://github.com/CodethinkLabs/mt3620-m4-samples/tree/master/SPI_RTApp_MT3620_BareMetal

Here is the link for the datasheet for the sensor on the ST-MK1160V1. In section 4.2, the first line of the table says that VDD max is 3.6 volts and VDD_IO max is .1v over VDD. This connection should be to H3 Pin 3, the 3.3v rail.

https://www.st.com/content/ccc/resource/technical/document/datasheet/a3/f5/4f/ae/8e/44/41/d7/DM00133076.pdf/files/DM00133076.pdf/jcr:content/translations/en.DM00133076.pdf

n7jti, 2020-05-01:
The power connection is correct in the I2C bare metal sample.

aiden-ct, 2020-05-13:
@n7jti - thanks for spotting this. Corrected in cdc1628f2eda4b7fe579ff4bfd176aa3d3652fb5.

closed 2020-05-13 by aiden-ct