CodethinkLabs/mt3620-m4-samples #7: Cortex-M4 Floating Point Unit (FPU)

mehrdadh, 2020-06-09:
Hi
One of the great features of M4 core is supporting FPU which could be used for signal processing or implementing different functionalities from [ARM CMSIS](https://github.com/ARM-software/CMSIS_5). I was wondering if there is any ongoing effort on showing how to enable FPU on M4? Maybe additional sample to your awesome samples.

I'm currently trying to do a simple example, but I ran into issues which I expressed here: https://github.com/Azure/azure-sphere-samples/issues/142

Hope to here from you soon,
Thanks

aiden-ct, 2020-06-15:
Hi @mehrdadh - sorry for the delay, I missed this in my inbox. I think the best place for you to get an answer is your Azure sphere samples issue. I'm not aware of any documentation for the Azure Sphere SDK that lays out how to enable the FPU. I'll keep an eye on your other issue, but if you could post your findings here, we can either add some instructions to the README or incorporate it into our samples as you suggest. Thanks for the issue!

mehrdadh, 2020-06-15:
@aiden-ct Thanks for getting back to me.
I think a sample would be very useful. I will keep you posted with my updates.

aiden-ct, 2020-07-10:
Solution left in related [issue](https://github.com/Azure/azure-sphere-samples/issues/142):

> Hi @mehrdadh and @aiden-ct - Sorry for the long wait. The floating point unit on M4 are turned off by default. Here are the steps to turn on this feature:
> 
>     Add this to the RT app code to turn on the floating point support on the M4 (as per these ARM instructions):
> 
> // Enable the FPU.
> // ARM 100166_0001_00_en S7.3.2
> __asm("ldr.w r0, =0xE000ED88"); // CPACR
> __asm("ldr r1, [r0]");
> __asm("orr r1, r1, 0xF << 20"); // [20:23] enable CP10 and CP11
> __asm("str r1, [r0]");  
> 
>     Add this to the RT app's CMakeLists.txt to get the compiler/linker to use this support:
> 
> target_compile_options(${PROJECT_NAME} PUBLIC -mfloat-abi=hard -mfpu=fpv4-sp-d16)
> target_link_libraries(${PROJECT_NAME} PUBLIC -lm -mfloat-abi=hard -mfpu=fpv4-sp-d16)
> 
> Please let me know if you have other questions.
> 
> Thanks,
> Johnny

aiden-ct, 2020-07-10:
@mehrdadh - can you verify that this is the fix? I'll try and find some time next week if not.

mehrdadh, 2020-07-12:
@aiden-ct I'll spend some time next week to try it. Thanks!

aiden-ct, 2020-09-10:
Related issue closed. Untested solution provided above.

closed 2020-09-10 by aiden-ct