CodethinkLabs/mt3620-m4-samples #12: SDCARD Write

taylor-an, 2020-08-31:
Hello,
Is there an example of writing on SD CARD?

ben-brewer-codethink, 2020-09-08:
Hey, there is an SD card example listed here: https://github.com/CodethinkLabs/mt3620-m4-samples/tree/master/SPI_SDCard_RTApp_MT3620_BareMetal

However it only reads/writes raw blocks, you'd need to use a fat32 or similar library to handle the filesystem.

taylor-an, 2020-09-09:
@ben-brewer-codethink 

Thank you for replying.
but there is only SD_ReadBlock() in SD.c. There is no SD_WriteBlock().

[SD.c line number 540](https://github.com/CodethinkLabs/mt3620-m4-samples/blob/master/SPI_SDCard_RTApp_MT3620_BareMetal/SD.c)
`bool SD_ReadBlock(const SDCard *card, uint32_t addr, void *data)`

ben-brewer-codethink, 2020-09-09:
Apologies, it's been a while since I looked into that code, you're correct.

Using the framework in SD.c it should be relatively straightforward to implement, the following document describes the process:
http://elm-chan.org/docs/mmc/mmc_e.html

aiden-ct, 2020-09-10:
@taylor-an - would you be able to push your write solution here as a PR at some point?

taylor-an, 2020-09-11:
> 
> 
> Apologies, it's been a while since I looked into that code, you're correct.
> 
> Using the framework in SD.c it should be relatively straightforward to implement, the following document describes the process:
> http://elm-chan.org/docs/mmc/mmc_e.html

Thank you.
It'll be helpful for me.

taylor-an, 2020-09-11:
> 
> 
> @taylor-an - would you be able to push your write solution here as a PR at some point?

OK, if I'll succeed.

aiden-ct, 2020-10-13:
@taylor-an - I added a write method to the SD card sample (see https://github.com/CodethinkLabs/mt3620-m4-samples/pull/15). Let me know if it's of any use to you.

closed 2020-10-14 by aiden-ct