CodethinkLabs/mt3620-m4-samples #14: Add Azure RTOS threadx support

xiongyu0523, 2020-10-10:
Can I submit a PR to add two Azure RTOS threadx samples? Here are my thoughts.

First one will be the classic demo_threadx projects for testing. I would like to submodule the official threadx 6.1 repo. 
The Second one is to demonstrate how to use async peripheral API together with RTOS semaphore and mutex to make it as a blocking API in a RTOS environment. 

Now I have a small 'blocker' when I'm making the first project. The default threadx cortex-m4 porting (.S) need two CMSIS compatible symbol 'PendSV_Handler' and 'SysTick_Handler'. I can directly modify VectorTable.c file in lib folder, but I want to make it as a submodule to be more consistent like other sample project. So I create a copy version under project root to do this.

Do you think it makes sense to rename all Exceptional handler name in VectorTable.c to align with CMSIS naming? 

```
extern const pFunc __Vectors[240];
       const pFunc __Vectors[240] __attribute__((used, section(".vectors"))) = {
  (pFunc)(&__StackTop),                     /*     Initial Stack Pointer */
  Reset_Handler,                            /*     Reset Handler */
  NMI_Handler,                              /* -14 NMI Handler */
  HardFault_Handler,                        /* -13 Hard Fault Handler */
  MemManage_Handler,                        /* -12 MPU Fault Handler */
  BusFault_Handler,                         /* -11 Bus Fault Handler */
  UsageFault_Handler,                       /* -10 Usage Fault Handler */
  0,                                        /*     Reserved */
  0,                                        /*     Reserved */
  0,                                        /*     Reserved */
  0,                                        /*     Reserved */
  SVC_Handler,                              /*  -5 SVCall Handler */
  DebugMon_Handler,                         /*  -4 Debug Monitor Handler */
  0,                                        /*     Reserved */
  PendSV_Handler,                           /*  -2 PendSV Handler */
  SysTick_Handler,                          /*  -1 SysTick Handler */

  /* Interrupts */
  ...
};
```

ben-brewer-codethink, 2020-10-12:
Hey, I think the easiest solution for now would be to use aliases to handle the handler function naming differences, see:
https://www.keil.com/support/man/docs/armcc/armcc_chr1359124980906.htm
https://gcc.gnu.org/onlinedocs/gcc-4.7.2/gcc/Function-Attributes.html

I have no strong preference about the vector table naming, only that to change it now might cause compatibility issues. If the aliasing approach can work then we could merge those changes to make future attempts to use CMSIS code easier.

xiongyu0523, 2020-10-13:
The problem is the arch porting file (.S) is within threadx repo and it hardcode to PendSV_Handler. And vector table are within mt3620-m4-drivers repo. If I want to submodule both without a change, I guess this alias attribute will NOT solve this problem?

```
    .global  PendSV_Handler
    .global  __tx_PendSVHandler
    .thumb_func
PendSV_Handler:
    .thumb_func
__tx_PendSVHandler:
@
@    /* Get current thread value and new thread pointer.  */
@
__tx_ts_handler:

#ifdef TX_ENABLE_EXECUTION_CHANGE_NOTIFY
@
@    /* Call the thread exit function to indicate the thread is no longer executing.  */
@
    CPSID   i                                       @ Disable interrupts
    PUSH    {r0, lr}                                @ Save LR (and r0 just for alignment)
    BL      _tx_execution_thread_exit               @ Call the thread exit function
```

xiongyu0523, 2020-10-13:
Also suggest we provide individual function implementation with weak attribute for all exceptions, even it is just while(1). Currently all faults go to `DefaultExceptionHandler()` and make difficulty for user to debug. 

Once have this changed, if a user is using a old naming PendSV() as or SysTick(), the code will go to a specific weak implementation and soon to correct the naming, or do whatever they like

ben-brewer-codethink, 2020-10-13:
Wait no, sorry not thought that through, we'd need a file, which has CMSIS stubs like:
```
// CMSIS Compatibility Stubs
void PendSV(void) { PendSV_Handler(); }
```

xiongyu0523, 2020-10-13:
emm... PendSV() and Systick() are heavily invoked exception handler during OS context switching.. Having this stub will degrade performance which count every cpu cycle. 

I can override by providing my own implementation of VectorTable.c for this single project if you think this is acceptable.

aiden-ct, 2020-10-13:
@xiongyu0523 - the problem is we don't want to cause compatibility issues for code that already depends on our drivers. If you could write a VectorTable.c with the threadx naming convention, then I think that would be best. We might want to split out the CPU_Freq stuff into it's own source file to minimise duplication.

xiongyu0523, 2020-10-14:
Sounds good. I will submit PR for Azure RTOS 6.1 port with standard demo_threadx project and wait for https://github.com/CodethinkLabs/mt3620-m4-drivers/issues/17 close to submit a I2C RTOS blocking example.

closed 2020-10-14 by xiongyu0523