CodethinkLabs/mt3620-m4-samples #19: IntercoreComms sample broken?

josmithua, 2020-12-08:
When debugging the IntercoreComms sample, the HL apps seems to 'pause' and doesn't log anymore, around a minute and a half after starting. There is no indication of any errors. Running on a device with OS version 20.10... (I've edited the CMakeLists.txt files)

I also noticed this is the only sample that has not been updated for 20.10. Is there a reason for that?

Thanks

aiden-ct, 2020-12-09:
Hi @josmithua - I'm unable to reproduce this. Can you provide more info? Maybe pause the execution in the debugger when you see the bug and inspect the call stack. I do remember seeing an issue where the sample would hang until i pressed button A or B on the dev board, but only saw it once early in the development and couldn't reproduce afterwards.

Thanks for highlighting the version mismatch for the IntercoreComms sample. I have updated master now to use 20.10 for that sample too (https://github.com/CodethinkLabs/mt3620-m4-samples/pull/20). Feel free to submit PRs in future if you spot any issues / want to upstream some of your improvements.

josmithua, 2020-12-09:
It seems I had some local changes that were causing issues that I thought I had discarded, but realized I hadn't when I went to pull your commits. Silly mistake on my part. 

I can confirm a _clean_ install of the sample works as expected :)

closed 2020-12-09 by josmithua

aiden-ct, 2020-12-09:
OK. No problem. Always nice to see people using our samples / drivers in the wild.