CodethinkLabs/mt3620-m4-samples #23: SDCARD Write/Read

taylor-an, 2021-01-05:
@aiden-ct  Thank you for commit to WriteBlock().

I tested at that time but I gave up to develop SDCARD Fat fs.
I trying to develop it again.

I tried to write and read blocks, from 0 to the end.
I used to SDCARD of Sandisk microSD HC1 class 4, 4GB capacity.
But I got too many errors to write and read till the end. The SPI clock was 250KHz, 2MHz, 20MHz.

When I got errors, I try to do SD_Close(), SPIMaster_Close() and SPIMaster_Open(), SD_Open() to retry to failed block. Well, it works.

But, I wonder is it right or not.
Is it possible to clearly(I mean without error)write and read all blocks?

And, if you can tell me the details of the SDCARD you tested.

aiden-ct, 2021-01-05:
@taylor-an - let me know how you get on and if my patch fixes things for you.

taylor-an, 2021-01-07:
@aiden-ct Thank you.
It works very well. I can't see any errors.

aiden-ct, 2021-01-07:
Great. Thanks for the bug report.

closed 2021-01-07 by aiden-ct