CodethinkLabs/mt3620-m4-samples #30: The UART library sets the baud rate incorrectly.

gtrevi, 2021-08-04:
Hi,
after doing some tests, it seems that the UART library sets baud rate incorrectly.

Repro:
Open the `UART_RTApp_MT3620_BareMetal` sample "as-is", change the baud rate for ISU0 to i.e. 9600 (main.c - line 135): the UART will still output at higher rate (i.e. about 300us/byte instead of about 980us/byte). This was confirmed by sampling the TX0 pin:

[image]

ben-brewer-codethink, 2021-08-06:
Hey,

I had a quick look at the UART code and it seems you're right and the baud calculation was wrong, I've created a fix here:
https://github.com/CodethinkLabs/mt3620-m4-drivers/commit/c4cdac1f63ed31d64586d15d5de0df3c441a20cb

Either re-clone the samples repository using:
git clone --recursive https://github.com/CodethinkLabs/mt3620-m4-samples.git -b benbrewer/uart-baud

Or point the submodule in your existing sample/project to the benbrewer/uart-baud branch in mt3620-m4-drivers

Please let me know if this resolves the issue for you.

gtrevi, 2021-08-06:
Hi @ben-brewer-codethink, I just tested the fix and it works, thank you!

closed 2021-08-09 by ben-brewer-codethink