CodethinkLabs/mt3620-m4-samples #31: Discrepancy between described  output  and actual output for IntercoreComms_Mailbox app

v-zzhe, 2021-11-04:
Following the instruction for IntercoreComms_Mailbox app, I got below output which is different from the  "Observe the output" section.

Actual output  when running the app from Visual Studio:
-------------------------------------------------
SocketEventHandler
Received 9 bytes: count-05.
Sending: hl-app-to-rt-app-adding00
SocketEventHandler
Received 9 bytes: count-05.
Sending: hl-app-to-rt-app-adding01
SocketEventHandler
Received 9 bytes: count-05.
Sending: hl-app-to-rt-app-adding02
SocketEventHandler
Received 9 bytes: count-05.
Sending: hl-app-to-rt-app-adding03

Actual output from serial terminal:
-------------------------------------------------
sending msg count-05
sending msg count-05
sending msg count-05
sending msg count-05
sending msg count-05
sending msg count-05
Message received: hl-app-to-rt-app-adding00
Sender: 25025d2c-66da-4448-bae1-ac26fcdd3627
sending msg count-05
Message received: hl-app-to-rt-app-adding01
Sender: 25025d2c-66da-4448-bae1-ac26fcdd3627
sending msg count-05
Message received: hl-app-to-rt-app-adding02
Sender: 25025d2c-66da-4448-bae1-ac26fcdd3627
sending msg count-05
Message received: hl-app-to-rt-app-adding03
Sender: 25025d2c-66da-4448-bae1-ac26fcdd3627
sending msg count-05

ben-brewer-codethink, 2021-11-12:
Hey, sorry for the late response on this, taking a look into this.

ben-brewer-codethink, 2021-11-15:
I've updated the documentation to match the current behaviour of the sample, sorry for the inconvenience.

closed 2021-11-15 by ben-brewer-codethink